mossdef-org/source.mossdef.org #179: [pbr] Issue: resolvers are unavailable despite installing all requirements

posita, 2023-01-14:
I see this on `/cgi-bin/luci/admin/services/pbr`:

> The _adguardhome.ipset_ is not supported on this system.
> The _dnsmasq.nftset_ is not supported on this system.
> Please check the [README](https://docs.openwrt.melmac.net/pbr/#use-resolvers-set-support) before changing this option.

I am running OpenWRT firmware version `OpenWrt 22.03.2 r19803-9a599fee93 / LuCI openwrt-22.03 branch git-22.288.45147-96ec0cd`.

Steps I took were pieced together from [the docs](https://docs.openwrt.melmac.net/pbr/#use-resolvers-set-support). (As an aside, this is no easy feat, since key information is scattered all over the place.)

From [this](https://docs.openwrt.melmac.net/pbr/#how-to-install-dnsmasq-full):

``` sh
opkg update
cd /tmp/ && opkg download dnsmasq-full
opkg remove dnsmasq
opkg install dnsmasq-full --cache /tmp/
rm -f /tmp/dnsmasq-full*.ipk
```

From [this](https://docs.openwrt.melmac.net/pbr/#requirements-pbr):

``` sh
opkg update; opkg install resolveip ip-full
```

From [this](https://docs.openwrt.melmac.net/pbr/#how-to-install-legacy-iptablesipset-packages):

``` sh
opkg update
opkg install ipset libnettle8 libnetfilter-conntrack3 iptables kmod-ipt-ipset iptables-mod-ipopt
```

From [this](https://docs.openwrt.melmac.net/pbr/#how-to-install---openwrt-2203-and-newer):

``` sh
opkg update
opkg install pbr luci-app-pbr
```

Finally:

``` sh
reboot
```

All operations were performed successfully.

stangri, 2023-01-14:
Please provide the output of `dnsmasq --version` and `AdGuardHome --version`.

posita, 2023-01-16:
Sorry for being unclear. I anticipated AdGuardHome being unavailable (since that's not installed), but `dnsmasq.nftset` was a surprise. Here's the `dnsmasq` version:

``` sh
root@...:~# dnsmasq --version
Dnsmasq version 2.86  Copyright (c) 2000-2021 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
```

stangri, 2023-01-16:
Evidently, installed dnsmasq-full does not support nft sets. Below is the output of dnsmasq with nft set support:

```
SG-105 in ~ # dnsmasq --version
Dnsmasq version 2.87  Copyright (c) 2000-2022 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
```

cjom, 2023-03-09:
Hi!
I am having this issue with
` AdGuard Home, version v0.108.0-b.29 `
and I suspect it is the version string compare that fails...

I am not running the opkg, I am running beta from /opt/AdGuardHome
Looking into the code I saw pbr expects to find /etc/AdGuardHome.yaml so I created a softlink and that seems to test true.
Also created a softlink in /usr/sbin/AdGuardHome and the "command -v AdGuardHome" also tests true.

So, could it be the version string compare? And if it is, could it be improved/fixed?

stangri, 2023-03-10:
What's the output of:
```sh
command -v AdGuardHome
$(command -v AdGuardHome) --version
$(command -v AdGuardHome) --version | sed 's|AdGuard Home, version v\(.*\)|\1|')
```

cjom, 2023-03-11:
I'm sorry, something was wrong before but not it is working.
I installed the adguarhome ipk and then replaced the executable with "edge" version. Just needed to tweak the init.d scripts.

Anyway, now PBR detects AdGuardHome.ipset. Thanks for all your time and work!
```
$(command -v AdGuardHome) --version | sed 's|AdGuard Home, version v\(.*\)|\1|'
0.108.0-a.462+a2053526
```

stangri, 2023-03-11:
`pbr 1.1.0-14` and later should parse the version string a bit better, however I'm still unclear where the executable and the yaml config files are for the manually installed AGH.

I'm closing this as with the opkg-installed version it's working as expected.

closed 2023-03-11 by stangri

cjom, 2023-03-12:
Yes, it is working.
When manually installing using the sh script, everything goes into the folder /opt/AdGuardHome except for init.d script that is named exactly like the folder: "AdGuardHome"