mossdef-org/source.mossdef.org #180: [pbr] issue: Inconsistent routing with PBR/OpenVPN

posita, 2023-01-16:
**Describe the bug**

I have a router configured as an OpenVPN client, and I use PBR to whitelist some domains and subdomains to be routed over my wan. I have noticed that after a period of time, the whitelisted domains and subdomains eventually revert to being routed over the VPN. It's unclear to me why/when this happens. I can almost always cure this by restarting `openvpn` and waiting a few seconds (maybe a minute tops). But it will eventually revert to trying to route those domains and subdomains over the VPN. I am not rebooting the router in the mean time, nor do I have any cron jobs or anything that periodically reset the VPN connection.

**Your configs**

1. ```/etc/config/dhcp```

```sh

config dnsmasq
	option domainneeded '1'
	option boguspriv '1'
	option filterwin2k '0'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option expandhosts '1'
	option nonegcache '0'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option nonwildcard '1'
	option localservice '1'
	option ednspacket_max '1232'
	option domain 'panda'
	option sequential_ip '1'
	list server '9.9.9.9'
	list server '149.112.112.112'
	list server '/airdns.org/52.48.66.85'
	list server '/airdns.org/54.225.156.17'
	option confdir '/tmp/dnsmasq.d'

config dhcp 'lan'
	option interface 'lan'
	option leasetime '12h'
	option dhcpv4 'server'
	option start '258'
	option limit '253'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

config domain
	option name 'modem'
	option ip '192.168.100.1'
   ```

2. ```/etc/config/network```

```sh

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'

config device
[... 286 lines cut ...]

Available commands:
        start           Start the service
        stop            Stop the service
        restart         Restart the service
        reload          Reload configuration files (or restart if service does not implement reload)
        enable          Enable service autostart
        disable         Disable service autostart
        enabled         Check if service is started on boot
        status          Generates output required to troubleshoot routing issues
                Use '-d' option for more detailed output
                Use '-p' option to automatically upload data under VPR paste.ee account
                        WARNING: while paste.ee uploads are unlisted, they are still publicly available
                List domain names after options to include their lookup in report
        version         Show version information
        on_firewall_reload      Run service on firewall reload
        on_interface_reload     Run service on indicated interface reload
        running         Check if service is running
        status          Service status
        trace           Start with syscall trace
        info            Dump procd service info

   ```

Content above has been slightly redacted to minimize identifying information.

posita, 2023-01-16:
Also, (FYI) I had to hand-correct the issue template formatting to get it to render correctly. The template has lines where code block entries are prefixed by white space which messes with things. E.g.:

```
    ```sh
This won't render properly.
    ```
```

stangri, 2023-01-16:
1. Your network file is missing VPN Client interface definition as recommended in the pbr README.
2. Which device are you running traceroute on?
3. Please provide the output of `service pbr status` in the failed routing state.
4.

stangri, 2023-01-16:
1. Your network file is missing VPN Client interface definition as recommended in the pbr README.
1. Which device are you running traceroute on?

Please provide the output of `service pbr status` in the failed routing state.

> If it's helpful for diagnostic purposes, I was experiencing similar issues with `vpn-policy-router`. (I know that's no longer being developed, but I mention it only in case it's a useful data point.)
> 
> Also, (FYI) I had to hand-correct the issue template formatting to get it to render correctly. The template has lines where code block entries are prefixed by white space which messes with things. E.g.:
> 
> ```
>     ```sh
> This won't render properly.
>     ```
> ```

Fixed in https://github.com/stangri/source.openwrt.melmac.net/commit/22cf54bc4cdc6e8477f7b8cece14773431fca8d7

stangri, 2023-01-16:
> If it's helpful for diagnostic purposes, I was experiencing similar issues with `vpn-policy-router`. (I know that's no longer being developed, but I mention it only in case it's a useful data [...]

Fixed in https://github.com/stangri/source.openwrt.melmac.net/commit/22cf54bc4cdc6e8477f7b8cece14773431fca8d7

posita, 2023-01-16:
> Your network file is missing VPN Client interface definition as recommended in the pbr README.

This [this](https://docs.openwrt.melmac.net/pbr/#basic-openvpn-client-config) the section to which you're referring? (There's a lot there and the organization is hard to navigate from a user perspective. I want to be sure I'm looking in the correct spot.)

> Which device are you running traceroute on?

A client machine connected to the router via its WiFi interface.

```
pbr 1.0.1-3 running on OpenWrt 22.03.2. WAN (IPv4): wan/wan/....
============================================================
Dnsmasq version 2.86  Copyright (c) 2000-2021 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
============================================================
Routes/IP Rules
default         10.xx.xx.1     128.0.0.0       UG    0      0        0 ovpnc0
default         ... 0.0.0.0         UG    0      0        0 wan
IPv4 table 256 route: default via ... dev wan 
IPv4 table 256 rule(s):
30000:	from all fwmark 0x10000/0xff0000 lookup pbr_wan
============================================================
Mangle IP Table: FORWARD
# Warning: iptables-legacy tables present, use iptables-legacy to see them
-N PBR_FORWARD
============================================================
Mangle IP Table: INPUT
# Warning: iptables-legacy tables present, use iptables-legacy to see them
-N PBR_INPUT
============================================================
Mangle IP Table: OUTPUT
# Warning: iptables-legacy tables present, use iptables-legacy to see them
-N PBR_OUTPUT
============================================================
Mangle IP Table: POSTROUTING
# Warning: iptables-legacy tables present, use iptables-legacy to see them
-N PBR_POSTROUTING
============================================================
Mangle IP Table: PREROUTING
# Warning: iptables-legacy tables present, use iptables-legacy to see them
-N PBR_PREROUTING
-A PBR_PREROUTING -d 192.168.100.1/32 -m comment --comment 192_168_100_1 -c 0 0 -g PBR_MARK_0x010000
-A PBR_PREROUTING -m set --match-set pbr_wan_4_dst_ip_cfg086ff5 dst -m comment --comment cds_audible_com -c 0 0 -g PBR_MARK_0x010000
-A PBR_PREROUTING -m set --match-set pbr_wan_4_dst_ip_cfg096ff5 dst -m comment --comment reg_usps_com -c 0 0 -g PBR_MARK_0x010000
============================================================
Mangle IP Table MARK Chain: PBR_MARK_0x010000
# Warning: iptables-legacy tables present, use iptables-legacy to see them
-N PBR_MARK_0x010000
-A PBR_MARK_0x010000 -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
-A PBR_MARK_0x010000 -c 0 0 -j RETURN
============================================================
Current ipsets
create pbr_wan_4_dst_ip_cfg076ff5 hash:ip family inet hashsize 1024 maxelem 65536 comment
create pbr_wan_4_dst_ip_cfg086ff5 hash:ip family inet hashsize 1024 maxelem 65536 comment
create pbr_wan_4_dst_ip_cfg096ff5 hash:ip family inet hashsize 1024 maxelem 65536 comment
============================================================
DNSMASQ sets
ipset=/cds.audible.com/pbr_wan_4_dst_ip_cfg086ff5 # cds.audible.com: cds.audible.com
ipset=/reg.usps.com/pbr_wan_4_dst_ip_cfg096ff5 # reg.usps.com: reg.usps.com
============================================================
Your support details have been logged to '/var/pbr-support'. [0;32m[✓][0m
```

posita, 2023-01-16:
Okay, thanks. Since the problem takes a few hours to manifest, can you give me a day or two to reset configs on my end and see if I still experience the issue? I'll close it or report back with an update.

posita, 2023-01-17:
I added the following to my `/etc/config/network` file:

```
config interface 'vpnclient'
	option proto 'none'
	option device 'ovpnc0'
```

I originally added `iface` instead of `device`, but when navigating to `.../cgi-bin/luci/admin/network/network`, I was informed that configuration would be automatically changed. The above is the result. (Maybe worth a doc change?)

`pbr` routing now seems more stable. However, if I ever restart `openvpn` (e.g., to reconnect to a different exit point), no whitelisted routing is possible until after I do `/etc/init.d/pbr restart` (or `reload`) as well (and even then, it can take quite awhile for the effects to be seen).

Closing, since the original issue _seems_ like it's resolved? I'll reopen if I discover otherwise.

stangri, 2023-01-17:
Does the firewall get restarted/reloaded when you restart OpenVPN? Sadly, OpenVPN does not create the netifd-compatible interfaces, so even when `pbr` tries to register to be notified of the OpenVPN interface changes, I don't think it ever does (get notified). In `vpn-policy-routing` I used a firewall hotplug to reload the service on OpenVPN reloads and I'm using a different script for firewall4. Is it no longer working with `pbr`?

posita, 2023-01-17:
Give me a bit to test again to make sure I'm observing the symptoms correctly.

posita, 2023-01-18:
After additional testing on my end, it looks like there's a lag, but eventually, `pbr` resets after `openvpn restart`. My observation(s) were probably a result of either: 1) not waiting long enough; or 2) caused by me mucking with other things trying to diagnose. I think this can be closed, since it does indeed appear that `pbr` is behaving as designed. Apologies for the false alarm!

closed 2023-01-18 by posita