mossdef-org/source.mossdef.org #182: [pbr] issue:

beatstick, 2023-03-14:
**Describe the bug**
I am normally using a vpn wireguard client as a default for all devices on the network.
I now have set up wan as default interface according to the faq because wan won't work when wginterface is disabled.
Now I need to reverse all rules but cannot chose wginterface when creating a new rule. Only wan is listed as an option for new rules.

I tried adding wginterface to the list of supported interfaces in the advaced luci config but it did not help.

**Your configs**

1. ```/etc/config/dhcp```

```sh

config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option localservice '1'
	option confdir '/tmp/dnsmasq.d'
	list addnhosts '/tmp/adb_list.overall'
	list server '127.0.0.1'
	list server '127.0.0.1#5453'
	list server '127.0.0.1#5453'
	list server '127.0.0.1'
	list server '127.0.0.1#5453'
	option dnsseccheckunsigned '1'
	option noresolv '1'
	option dnssec '1'

config dhcp 'lan'
	option interface 'lan'
	option leasetime '12h'
	option dhcpv6 'server'
	option ra 'server'
	option ra_management '1'
	option start '130'
	option limit '180'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

```
...
static routes

3. ```/etc/config/network```

```sh
config interface 'loopback'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'
	option device 'lo'

config globals 'globals'
	option ula_prefix 'fdad:e1b4:7261::/48'

config interface 'lan'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option ipaddr 'xxx'
	option dns '9.9.9.9 8.8.8.8'
	option device 'br-lan'

config interface 'wan'
	option proto 'dhcp'
	option peerdns '0'
	option device 'eth1'
	option dns '127.0.0.1'

config interface 'wan6'
	option proto 'dhcpv6'
	option reqprefix 'auto'
	option reqaddress 'try'
	option peerdns '0'
	option device 'eth1'
	option dns '0::2'

config interface 'WGINTERFACE'
	option proto 'wireguard'
	list addresses 'xxx'
	option private_key 'xxx'

config wireguard_WGINTERFACE
	option persistent_keepalive '25'
	option route_allowed_ips '0'
	list allowed_ips '0.0.0.0/0'
	list allowed_ips '::/0'
	option endpoint_port '51821'
	option endpoint_host 'xxx'
	option public_key 'xxx'
	option description 'xxx'

config device
	list ports 'eth0'
	option type 'bridge'
	option name 'br-lan'

```

4. ```/etc/config/firewall```

```sh
config defaults
	option syn_flood '1'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'REJECT'

config zone
	option name 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option network 'lan'

config zone
	option name 'wan'
	option output 'ACCEPT'
	option masq '1'
	option mtu_fix '1'
	option input 'DROP'
	option forward 'DROP'
	list network 'wan'
	list network 'wan6'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'DROP'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option src_ip 'fc00::/6'
	option dest_ip 'fc00::/6'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
	list icmp_type '131/0'
	list icmp_type '132/0'
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'

config include
	option path '/etc/firewall.user'

config zone
	option output 'ACCEPT'
	option name 'WGZONE'
	option masq '1'
	option mtu_fix '1'
	option input 'DROP'
	option forward 'DROP'
	list network 'WGINTERFACE'

config include
	option path '/etc/firewall.cs'
	option enabled '1'
	option reload '1'

	option fw4_compatible '1'
	option type 'script'
	option path '/usr/share/pbr/pbr.firewall.include'

plus some local network port forwards

```

5. ```/etc/config/pbr```

default config at the moment.

```sh

```

**To Reproduce**
Steps to reproduce the behavior:
1. enable wan as default route
2. use wginterface for policy based rules

**Expected behavior**
 wginterface should be a valid interface to route via.

**Policy Routing run-time information**

1. Output of ```/etc/init.d/pbr reload``` with ```verbosity``` set to 2:

```sh
 /etc/init.d/pbr reload
Activating traffic killswitch [✓]
Setting up routing for 'wan/eth1/xxx' [✓]
Setting up routing for 'WGINTERFACE/xxx' [✓]
Deactivating traffic killswitch [✓]
pbr 1.1.0-15 monitoring interfaces: wan WGINTERFACE 
pbr 1.1.0-15 (nft) started with gateways:
wan/eth1/xxx [✓]
WGINTERFACE/xxx

```

2. Output of ```/etc/init.d/pbr status```:

```sh
pbr - environment
pbr 1.1.0-15 running on OpenWrt 22.03.3. WAN (IPv4): wan/eth1/31.16.82.254.
============================================================
Dnsmasq version 2.86  Copyright (c) 2000-2021 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
============================================================
pbr chains - policies
	chain pbr_forward { # handle 40
	}
	chain pbr_input { # handle 41
	}
	chain pbr_output { # handle 42
	}
	chain pbr_prerouting { # handle 43
	}
	chain pbr_postrouting { # handle 44
	}
============================================================
pbr chains - marking
	chain pbr_mark_0x010000 { # handle 796
		counter packets 0 bytes 0 meta mark set meta mark & 0xff01ffff | 0x00010000 # handle 797
		return # handle 798
	}
	chain pbr_mark_0x020000 { # handle 799
		counter packets 0 bytes 0 meta mark set meta mark & 0xff02ffff | 0x00020000 # handle 800
		return # handle 801
	}
============================================================
pbr nft sets
============================================================
IPv4 table 255 route: 
IPv4 table 255 rule(s):
0:	from all lookup local
IPv4 table 256 route: default via xxx dev WGINTERFACE 
IPv4 table 256 rule(s):
30001:	from all fwmark 0x20000/0xff0000 lookup pbr_WGINTERFACE

```

stangri, 2023-03-14:
what's the version of luci-app-pbr? Please clear the browser cache and try again.

beatstick, 2023-03-14:
opkg list  luci-app-pbr
luci-app-pbr - 1.1.0-15 - Provides Web UI for Policy Based Routing Service.
luci-app-pbr - 1.1.0-1 - Provides Web UI for Policy Based Routing Service.

via luci: Running (version: 1.1.0-15 using nft)

stangri, 2023-03-14:
You need to update the luci app to be on the same version as the main package, then clear browser cache.

beatstick, 2023-03-14:
opkg list-installed luci-app-pbr
luci-app-pbr - 1.1.0-15

opkg list-installed pbr
pbr - 1.1.0-15

They are the same. Clearing browser cache did not help unfortunately.

beatstick, 2023-03-14:
I've now assigned the wireguard interface to the firewall-zone of the wan, and I can now disable the wireguard interface and the connection still works. 
So it now should work as intended with the pbr tool. I can now set wginterface as route, but now I need the old rules again. And unfortunately if I set my pcs ip to use the wan via pbr rule, it stll connects to the internet via wireguard client...

I should try solving this tomorrow...

stangri, 2023-03-14:
What's the output of `ubus -S call luci.pbr getInterfaces '{"name": "pbr" }'`?

It may be that there's an issue with capitalized interface names.

beatstick, 2023-03-14:
ubus -S call luci.pbr getInterfaces '{"name": "pbr" }'
{"pbr":{"interfaces":["wan","WGINTERFACE"]}}

stangri, 2023-03-14:
I'd try to reboot the router to see if that helps with this issue. If you have any other concerns, please open a separate issue.

beatstick, 2023-03-14:
I think I am going back to using vpn-policy-routing for now, which seems to to sometimes produce errors on openwrt 22.03, but at least it mostly works...

beatstick, 2023-03-21:
This issue can be closed. It started to work after reinstalling pbr.

closed 2023-10-24 by stangri