mossdef-org/source.mossdef.org #183: pbr service error: failed to set up interfaces

beatstick, 2023-03-21:
The service pbr sometimes fails to set up any interface. 
I get the error message: "Failed to set up any interface" on the gui.

logread -e pbr

Tue Mar 21 11:03:52 2023 user.notice pbr: Reload on interface status aborted: service not running.
Tue Mar 21 11:03:52 2023 user.notice pbr: Reload on interface status aborted: service not running.
Tue Mar 21 11:03:53 2023 user.notice pbr: Reload on interface status aborted: service not running.
Tue Mar 21 11:03:53 2023 user.notice pbr: Reloading pbr due to firewall action: includes
Tue Mar 21 11:03:53 2023 user.notice pbr: Reload on firewall action aborted: service not running.
Tue Mar 21 11:03:53 2023 user.notice pbr: Reloading wan due to ifup of wan (eth1)
Tue Mar 21 11:03:55 2023 user.notice pbr: Reloading pbr due to firewall action: includes
Tue Mar 21 11:03:55 2023 user.notice pbr: Reload on firewall action aborted: service not running.
Tue Mar 21 11:03:55 2023 user.notice pbr: Reloading WGINTERFACE due to ifup of WGINTERFACE (WGINTERFACE)
Tue Mar 21 12:33:19 2023 user.notice pbr: Reload on interface status aborted: service not running.
Tue Mar 21 12:33:20 2023 user.notice pbr: Reloading pbr due to firewall action: includes
Tue Mar 21 12:33:20 2023 user.notice pbr: Reload on firewall action aborted: service not running.
Tue Mar 21 12:33:20 2023 user.notice pbr: Reloading wan due to ifupdate of wan (eth1)
Tue Mar 21 13:48:04 2023 user.notice pbr: Reload on interface status aborted: service not running.
Tue Mar 21 13:48:06 2023 user.notice pbr: Reloading pbr due to firewall action: includes
Tue Mar 21 13:48:06 2023 user.notice pbr: Reload on firewall action aborted: service not running.
Tue Mar 21 13:48:06 2023 user.notice pbr: Reloading wan due to ifupdate of wan (eth1)
Tue Mar 21 15:17:48 2023 user.notice pbr: Reload on interface status aborted: service not running.
Tue Mar 21 15:17:49 2023 user.notice pbr: Reloading pbr due to firewall action: includes
Tue Mar 21 15:17:50 2023 user.notice pbr: Reload on firewall action aborted: service not running.
Tue Mar 21 15:17:50 2023 user.notice pbr: Reloading wan due to ifupdate of wan (eth1)
Tue Mar 21 16:32:34 2023 user.notice pbr: Reload on interface status aborted: service not running.
Tue Mar 21 16:32:35 2023 user.notice pbr: Reloading pbr due to firewall action: includes
Tue Mar 21 16:32:35 2023 user.notice pbr: Reload on firewall action aborted: service not running.
Tue Mar 21 16:32:35 2023 user.notice pbr: Reloading wan due to ifupdate of wan (eth1)

beatstick, 2023-03-22:
-----

cat /etc/config/dhcp

config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option localservice '1'
	option confdir '/tmp/dnsmasq.d'
	list addnhosts '/tmp/adb_list.overall'
	list server '127.0.0.1'
	list server '127.0.0.1#5453'
	list server '127.0.0.1#5453'
	list server '127.0.0.1'
	list server '127.0.0.1#5453'
	option dnsseccheckunsigned '1'
	option noresolv '1'
	option dnssec '1'

config dhcp 'lan'
	option interface 'lan'
	option leasetime '12h'
	option dhcpv6 'server'
	option ra 'server'
	option ra_management '1'
	option start '130'
	option limit '180'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

cat /etc/config/firewall

config defaults
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option synflood_protect '1'

config zone
	option name 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option network 'lan'

config zone
	option name 'wan'
	option output 'ACCEPT'
	option masq '1'
	option mtu_fix '1'
	option input 'DROP'
	option forward 'DROP'
	list network 'wan'
	list network 'wan6'
[... 629 lines cut ...]
/etc/init.d/pbr reload

Activating traffic killswitch [✓]
Setting up routing for 'wan/eth1/xxx/fe80::2a0:ff:fec0:1103/64' RTNETLINK answers: File exists
[✓]
Setting up routing for 'WGINTERFACE/xxx/::/0' RTNETLINK answers: File exists
RTNETLINK answers: File exists
[✓]
Routing 'mediathek' via wan [✓]
Routing 'nextcloud talk' via wan [✓]
Routing 'searx' via wan [✓]
Routing 'jitsi-meet' via wan [✓]
Routing 'Aurora Store no vpn' via wan [✓]
Routing 'elden ring ports' via wan [✓]
Routing 'sniper elite 5' via wan [✓]
Routing 'debian server' via wan [✓]
Routing 'bloody trapland ' via wan [✓]
Routing 'steftop no vpn' via wan [✓]
Routing 'stefdroid old no vpn' via wan [✓]
Routing 'ebay kleinanzeigen' via wan [✓]
Deactivating traffic killswitch [✓]
pbr 1.1.0-19 monitoring interfaces: wan WGINTERFACE 
pbr 1.1.0-19 (nft) started with gateways:
wan/eth1/xxx/xxx:1103/64
WGINTERFACE/xxx/::/0 [✓]

stangri, 2023-03-23:
While I haven't figured out what's producing the `RTNETLINK answers: File exists` (I suspect it's IPv6 related), the general start up should have been addressed in `pbr 1.1.0-22` available from [my dev repo](https://dev.melmac.net/repo/). I'd appreciate if you test it and let me know.

beatstick, 2023-03-23:
Ok, I installed it and will report back if the error still occurs.

This is how the install went:

opkg install pbr_1.1.0-22_all.ipk
Upgrading pbr on root from 1.1.0-19 to 1.1.0-22...
Stopping pbr service... 
Activating traffic killswitch [✓]
Removing routing for 'wan/eth1/xxx/64' [✓]
Removing routing for 'WGINTERFACE/xxx/::/0' [✓]
Deactivating traffic killswitch [✓]
pbr 1.1.0-19 (nft) stopped [✓]
OK
Removing rc.d symlink for pbr... OK
Command failed: Not found
Configuring pbr.
Installing rc.d symlink for pbr... OK
uci: Entry not found
uci: Entry not found

uci: Entry not found

uci: Entry not found

uci: Entry not found

uci: Entry not found

uci: Entry not found

Activating traffic killswitch [✓]
Setting up routing for 'wan/eth1/xxx1103/64' RTNETLINK answers: File exists
[✓]
Setting up routing for 'WGINTERFACE/xxx/::/0' RTNETLINK answers: File exists
RTNETLINK answers: File exists
[✓]
Routing 'mediathek' via wan [✓]
Routing 'nextcloud talk' via wan [✓]
Routing 'searx' via wan [✓]
Routing 'jitsi-meet' via wan [✓]
Routing 'Aurora Store no vpn' via wan [✓]
Routing 'elden ring ports' via wan [✓]
Routing 'sniper elite 5' via wan [✓]
Routing 'debian server' via wan [✓]
Routing 'bloody trapland ' via wan [✓]
Routing 'steftop no vpn' via wan [✓]
Routing 'stefdroid old no vpn' via wan [✓]
Routing 'ebay kleinanzeigen' via wan [✓]
Deactivating traffic killswitch [✓]
pbr 1.1.0-22 monitoring interfaces: wan WGINTERFACE 
pbr 1.1.0-22 (nft) started with gateways:
wan/eth1/xx/64
WGINTERFACE/1xxx/::/0 [✓]

beatstick, 2023-03-25:
No problems so far with pbr 1.1.0-22. The system has been running stable for the last two days and the interfaces always stay up.

stangri, 2023-03-27:
Closing for now, feel free to reopen if you still experience the issue.