mossdef-org/source.mossdef.org #184: [pbr] wish: interface specific rule reload on interface restart - wireguard interface restart causes pbr to re-apply rules for all interfaces

dieskim, 2023-06-13:
**Is your feature request related to a problem? Please describe.**
We run multiple wireguard interfaces on our openwrt install. We run watchcat to monitor these interface. When ever one of these interfaces looses connection with its server watchcat will restart the interface. This causes pbr to reload ALL its rules. Thus causes all the devices connected via the other interfaces to also loose there connection until pbr reloads and applies all its rules.

**Describe the solution you'd like**
It would be great if pbr could only reload the specific rule for the specific interface rather than all its rules. 

**Describe alternatives you've considered**
Maybe there is a way to do this that I missed? If so please let me know, and Sorry!

**Additional context**
Thanks for the awesome tool. Its working well for us with multiple wireguard and multipe devices.

stangri, 2023-06-13:
There are many issues with quick-reloads (just rebuilding the routing table for the affected interface without resetting anything else). I took a stab at addressing them in the last builds of `vpn-policy-routing` and then yet again when migrating to `pbr`.

It may be a multi-step process to achieve this, I hope to see some improvements in the coming builds of `pbr`.

dieskim, 2023-06-13:
Thanks for the info. Trying to think of a solution for us, could we just disable the pbr reload on the wg interface reload? Assuming the wg interface is only restarting I think the gateway info should not be changing, right? But we would need to reload pbr to rebuild the routing tables if we change the wg interface info etc ?

dieskim, 2023-06-14:
I also noticed the following, on a interface restart it seems to trigger pbr to reload multiple times, is this necessary?

It seems to reload in the following order when a interface restarts:
1. interface restart causes the firewall to reload (via etc/hotplug.d/iface/20-firewall) - this causes the pbr method on_firewall_reload to run on_start
2. interface restart then causes pbr method on_interface_reload to run on_interface_reload
3. if you have the hotplug script installed (/etc/hotplug.d/iface/70-pbr) this then causes it to reload again via pbr method reload_interface

I am not sure its needed to reload all these times? Sometimes this can take a while to reload all these times.

stangri, 2023-06-23:
Well, when the firewall is reloaded, pbr needs to reload the policies. When interface is reloaded, **ideally** only the relevant routing table for that interface needs to be reloaded. 

Long-term plan is to migrate to atomic nft/nft-file support that way the firewall reload will not trigger pbr service reload.

dieskim, 2023-06-27:
Thanks for the info.

I am just wondering if some chaining happening that could maybe be avoided, where an Interface Restart causes:
on_firewall_reload
but also
on_interface_reload
and also
reload_interface

all 3 of these cause PBR to reload the policies, 3 times, maybe it could only reload once? 

In my case PBR loops over all this and reloads its policies multiple times - with 10 interfaces it sometimes gets into a loop and reloads all the policies like 30 times - this takes a while to finish.

But maybe this is necessary. 

Thanks again for the great tool. Happy to help and test where I can.

stangri, 2024-05-13:
The most recent version from this repo can be ran with `nft_file_mode` option set to 1 which will only reload actual policies when needed and it only reloads a single interface routing table on the update.

Please reopen if this fails to meet your expectations.

closed 2024-05-13 by stangri