mossdef-org/source.mossdef.org #185: [pbr] issue: timeout waiting for wan gateway on USB-based WAN

p00rt, 2023-08-10:
**Describe the bug**
To connect my router to the internet I either connect a smartphone to it and run USB tethering or connect to another network via WiFi. When I'm connected via wwan pbr works as expected, but when I switch over to USB it doesn't come up - instead it prints `pbr 1.1.1-7 waiting for wan gateway...` a bunch of times until it timeouts.

**Your configs**

1. ```/etc/config/dhcp```

```sh
config dnsmasq
	option domainneeded '1'
	option boguspriv '1'
	option filterwin2k '0'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option nonegcache '0'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option nonwildcard '1'
	option localservice '1'
	option ednspacket_max '1232'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	option dhcpv6 'server'
	option ra 'server'
	option ra_slaac '1'
	list ra_flags 'managed-config'
	list ra_flags 'other-config'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'
```

2. ```/etc/config/network```
```sh
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd7f:6caf:b381::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0.1'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config switch
[... 296 lines cut ...]
pbr 1.1.1-7 running on OpenWrt 22.03.5. WAN (IPv4): wwan//0.0.0.0.
============================================================
Dnsmasq version 2.86  Copyright (c) 2000-2021 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
============================================================
pbr chains - policies
	chain pbr_forward { # handle 36
	}
	chain pbr_input { # handle 37
	}
	chain pbr_output { # handle 38
	}
	chain pbr_prerouting { # handle 39
	}
	chain pbr_postrouting { # handle 40
	}
============================================================
pbr chains - marking
============================================================
pbr nft sets
============================================================
IPv4 table 257 route: 
IPv4 table 257 rule(s):
30001:	from all fwmark 0x20000/0xff0000 lookup pbr_usb
```

stangri, 2023-08-10:
When internet connection via cellphone/usb is active, could you please run the commands below and provide me with the output:
```sh
. /lib/functions/network.sh 
network_find_wan wan
echo "$wan"
```

p00rt, 2023-08-11:
Thanks for your incredibly quick reply! Here's the output
```sh
root@OpenWrt:~# . /lib/functions/network.sh 
root@OpenWrt:~# network_find_wan wan
root@OpenWrt:~# echo "$wan"
hu_bud_wg_101
root@OpenWrt:~# ls -la "/sys/devices/virtual/net/usb0/tun_flags"
ls: /sys/devices/virtual/net/usb0/tun_flags: No such file or directory 
```
there is in fact no `/sys/devices/virtual/net/usb0` directory at all.

stangri, 2023-08-11:
In the newer versions available from my repo, you can set the `procd_wan_interface` option to `usb` and it will use the `usb` as the wan interface. You'd have to script setting this option when you switch to phone as your gate to the internet and remove it when you switch back to `wwan`.  You can also try renaming `usb` to `wanusb` and `pbr` may discover it as a `wan` interface then.

Outside of that, there's no way really for `pbr` to discover that `usb` is your `wan` interface.

stangri, 2024-05-13:
Closing due to lack of replies.

closed 2024-05-13 by stangri