mossdef-org/source.mossdef.org #193: [pbr] issue: PBR intermittently ineffective after a few hours and all traffic routed over VPN

posita, 2024-01-18:
**Describe the bug**
I have a router configured to connect three LANs (`lan`, `lan2`, and `guest`) to two OpenVPN clients (`vpn0` and `vpn1`). I use `pbr` to whitelist some domains and subdomains to be routed over my WAN or my second VPN. Since [upgrading `pbr` recently (from about a year ago)](https://forum.openwrt.org/t/policy-based-routing-pbr-package-discussion/140639/808?u=eb3f73) and [changing my `resolve_set` argument from `dnsmasq.ipset` to `dnsmasq.nftset`](https://forum.openwrt.org/t/policy-based-routing-pbr-package-discussion/140639/814?u=eb3f73) after domain resolution stopped working, it seems `pbr` now stops having an effect after a short time (1-2 hours), and only rebooting the router (sometimes `/etc/init.d/network restart`) will help. Restarting `pbr` does not. It's unclear to me why/when this happens.

**Your configs**

1. ```/etc/config/dhcp```

```sh
config dnsmasq
	option domainneeded '1'
	option boguspriv '1'
	option filterwin2k '0'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'xxx'
	option expandhosts '1'
	option nonegcache '0'
	option cachesize '1000'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option nonwildcard '1'
	option localservice '1'
	option ednspacket_max '1232'
	option filter_aaaa '0'
	option filter_a '0'
	option sequential_ip '1'
	list server '9.9.9.9'
	list server '149.112.112.112'
	option confdir '/tmp/dnsmasq.d'

config dhcp 'lan'
	option interface 'lan'
	option start '258'
	option limit '253'
	option leasetime '12h'
	option dhcpv4 'server'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

config dhcp 'vpn1'
	option interface 'vpn1'
	option start '258'
	option limit '253'
	option leasetime '12h'

config dhcp 'guest'
	option interface 'guest'
	option start '258'
	option limit '253'
	option leasetime '12h'
	option force '1'
	list dhcp_option '6,9.9.9.9'
	list dhcp_option '6,149.112.112.112'
```

2. ```/etc/config/network```

```sh
[... 447 lines cut ...]
		type ipv4_addr
		flags interval
		counter
		auto-merge
		comment "hulu-wan"
	}
============================================================
dnsmasq sets
nftset=/espn.api.edge.bamgrid.com/4#inet#fw4#pbr_wan_4_dst_ip_cfg0b6ff5 # espn-wan
nftset=/d2f2ekwwtg17a.cloudfront.net/4#inet#fw4#pbr_wan_4_dst_ip_cfg0b6ff5 # espn-wan
nftset=/espn.com/4#inet#fw4#pbr_vpn1_4_dst_ip_cfg0c6ff5 # espn-vpn
nftset=/live-mediashield-hulustream-com.akamaized.net/4#inet#fw4#pbr_wan_4_dst_ip_cfg0d6ff5 # hulu-wan
nftset=/hulu.com/4#inet#fw4#pbr_wan_4_dst_ip_cfg0d6ff5 # hulu-wan
nftset=/live-f.hulustream.com/4#inet#fw4#pbr_wan_4_dst_ip_cfg0d6ff5 # hulu-wan
============================================================
IPv4 table 256 route: default via 73.177.1.1 dev wan 
IPv4 table 256 rule(s):
30000:	from all fwmark 0x10000/0xff0000 lookup pbr_wan
IPv4 table 257 route: default via 10.32.154.7 dev ovpnc0 
IPv4 table 257 rule(s):
30001:	from all fwmark 0x20000/0xff0000 lookup pbr_vpn0
IPv4 table 258 route: default via 10.7.2.3 dev ovpnc1 
IPv4 table 258 rule(s):
30002:	from all fwmark 0x30000/0xff0000 lookup pbr_vpn1
```

posita, 2024-01-18:
I made some tweaks to my `firewall` and `pbr`, and now things _**appear**_ to have stabilized, so I'm going to close this to avoid unproductively consuming your bandwidth. I'll reopen if I see it again.

stangri, 2024-01-18:
Thanks for being proactive on this. What's the reason for the `espn-vpn` policy? My hunch is that's where the problem is. Maybe the `espn.api.edge.bamgrid.com` matches some higher level espn.com domain which eventually gets added to the set and the routing switches to VPN.

Also, if you're already on the full nft version you may want to switch to a newer `pbr` from my repo which supports (and uses by default) the fw4-compatible nft file.

posita, 2024-01-21:
Reopening because this is still an issue for me.

> Thanks for being proactive on this. What's the reason for the `espn-vpn` policy? My hunch is that's where the problem is. Maybe the `espn.api.edge.bamgrid.com` matches some higher level espn.com domain which eventually gets added to the set and the routing switches to VPN.

I'm not sure I understand this. Are you saying that there is overlap of IPs among `espn.api.edge.bamgrid.com` (which maps to `d2f2ekwwtg17a.cloudfront.net`) and something in the `espn.com` domain? How would that explain that **_all_** traffic (including that going to `hulu.com`) is routed via `vpn0`?

> Also, if you're already on the full nft version you may want to switch to a newer `pbr` from my repo which supports (and uses by default) the fw4-compatible nft file.

I tried following your advice and now appear to be in a worse spot. More specifically, I added your repo per [these instructions](https://docs.openwrt.melmac.net/#add-repository-github). Then I did `opkg update ; opkg upgrade pbr luci-app-pbr`. This resulted in an error, and now `pbr` appears not to work at all.

```
# opkg upgrade luci-app-pbr pbr
Upgrading luci-app-pbr on root from 1.1.1-7 to 1.1.3-25...
Downloading https://repo.openwrt.melmac.net/luci-app-pbr_1.1.3-25_all.ipk
Upgrading pbr on root from 1.1.1-7 to 1.1.3-25...
Downloading https://repo.openwrt.melmac.net/pbr_1.1.3-25_all.ipk
Stopping pbr service...
Activating traffic killswitch [✓]
Removing routing for 'wan/<isp-gateway-ip>' [✓]
Removing routing for 'vpn0/ovpnc0/<vpn0-gateway-ip>' [✓]
Removing routing for 'vpn1/ovpnc1/<vpn1-gateway-ip>' [✓]
Deactivating traffic killswitch [✓]
pbr 1.1.1-7 (nft) stopped [✓]
OK
Removing rc.d symlink for pbr... OK
Command failed: Not found
Removing obsolete file /etc/hotplug.d/iface/70-pbr.
Removing obsolete file /usr/share/pbr/pbr.firewall.include.
Configuring pbr.
Installing rc.d symlink for pbr... OK
Setting up routing for 'wan/<isp-gateway-ip>' [✓]
Setting up routing for 'vpn0/ovpnc0/<vpn0-gateway-ip>' [✓]
Setting up routing for 'vpn1/ovpnc1/<vpn1-gateway-ip>' [✓]
Routing 'espn-wan' via wan [✓]
Routing 'espn-vpn' via wan [✓]
Routing 'hulu-wan' via wan [✓]
Installing fw4 nft file [✗]
Restarting dnsmasq [✓]
pbr 1.1.3-25 monitoring interfaces: wan vpn0 vpn1
pbr 1.1.3-25 (nft mode) started with gateways:
wan/<isp-gateway-ip>
vpn0/ovpnc0/<vpn0-gateway-ip> [✓]
vpn1/ovpnc1/<vpn1-gateway-ip>
ERROR: Failed to install fw4 nft file '/var/run/pbr.nft'!
Configuring luci-app-pbr.
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/pbr is different from the conffile in the new package. The new conffile will be placed at /etc/config/pbr-opkg.
 ```

stangri, 2024-01-21:
The only reason I've seen nft file fail to install is the IP overlap which nft doesn't handle gracefully. You can make the new version work same as before by setting `nft_file_support` option to 0.

posita, 2024-01-21:
Okay, I think these came from me fiddling with stuff between my original observation of this issue. I filed #194 as a result of that investigation. I've now removed/corrected my entries to comply with the work-around described in that issue, so no more `nft` error. After upgrading and rebooting my router, routing _**appears**_ to work as desired, until I do this:

```
/etc/init.d/pbr stop
# wait for completion
/etc/init.d/openvpn restart
# manually wait for "Initialization Sequence Completed" in logs and verify all traffic now routed to `vpn0`
/etc/init.d/pbr start
# wait for completion
```

From that point on, all traffic is routed via `vpn0` indefinitely and `pbr` does not appear to affect any routing whatsoever.

stangri, 2024-01-21:
```sh
/etc/init.d/pbr stop
# wait for completion
/etc/init.d/network restart
# wait for completion; now all traffic is routed through `wan`
/etc/init.d/openvpn restart
# manually wait for "Initialization Sequence Completed" in logs and verify all traffic now routed to `vpn0`
/etc/init.d/pbr start
# wait for completion
```

What's the output of the start command you ran last and the `service pbr status` at this point?

mtompkins, 2024-02-19:
I'm see this as well. Will revert with some data shortly.

mtompkins, 2024-02-29:
Here's what I've found, but please understand it is preliminary: I have the following
Running (version: 1.1.1-7 using iptables).

WAN (ISP)
- VPN1 (Wireguard)  Default Gateway
- VPN2 (OpenVPN)

On router reboot, the OpenVPN takes extended time to spin up. 
There are rules within PBR for this egress.

What I am finding is, pbr doesn't work until two things occur. 
That VPN2 has to come up, and I have to manually restart pbr.

I know this isn't the most helpful, but I'm trying to work into it as time allows.

stangri, 2024-02-29:
@mtompkins this issue is about pbr not working a few hours after start, you're mentioning something about immediately after reboot, those are not the same issues. If you want me to have a look at what's happening on your system, then:
1. Use the latest pbr from my repo
2. open a new issue
3. include information requested in the Getting Help section of the README

posita, 2025-01-14:
Closing, since this is stale, but I'm still fighting issues. (See #207.)

closed 2025-01-14 by posita