mossdef-org/source.mossdef.org #201: [pbr] issue: Unexpected exit code -9 on firewall reload

schuettecarsten, 2024-05-28:
When firewall is reloaded, pbr is executed to update routes. That's fine so far. But the script exits with exit code -9 before pbr finished to update the rules, causing the firewall reload script to end before pbr finished to update. That might cause race conditions if firewall is reloaded multiple times.

```
root@GatewayDummi:/etc/init.d# ./firewall reload
Automatically including '/usr/share/nftables.d/table-post/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/table-post/30-pbr.nft'
Automatically including '/usr/share/nftables.d/chain-post/dstnat/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/forward/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/mangle_forward/30-pbr.nft'
Automatically including '/usr/share/nftables.d/chain-post/mangle_input/30-pbr.nft'
Automatically including '/usr/share/nftables.d/chain-post/mangle_output/30-pbr.nft'
Automatically including '/usr/share/nftables.d/chain-post/mangle_postrouting/30-pbr.nft'
Automatically including '/usr/share/nftables.d/chain-post/mangle_prerouting/30-pbr.nft'
Automatically including '/usr/share/nftables.d/chain-post/srcnat/20-miniupnpd.nft'
Setting up routing for 'wan/pppoe-wan/***.80/***:d082/64' [✓]
Setting up routing for 'vpn_wg0/***.1/***::1/64' [✓]
Setting up routing for 'vpn_wg1/***.1/***::1/124' [✓]
Routing 'ssh' via vpn_wg1 [✓]
Routing 'ssh_v6' via vpn_wg1 [✓]
Routing 'rdp' via vpn_wg1 Include '/usr/share/pbr/firewall.include' failed with exit code -9
root@GatewayDummi:/etc/init.d# [✓]
Routing 'rdp_v6' via vpn_wg1 [✓]
pbr 1.1.4-16 monitoring interfaces: wan vpn_wg0 vpn_wg1

```

schuettecarsten, 2024-05-28:
Oh yes,  that looks better.

schuettecarsten, 2024-06-03:
IÄve just reopened this issue. The nft file is created at `'/usr/share/nftables.d/`, but these files are not included in the list of files that should be persisted on firmware upgrade. No on the first boot after a firmware flash, the files are missing and it looks like they're not recreated automatically?

stangri, 2024-06-03:
Seems to me like different issues tho.

Could you please elaborate which files are missing after first boot and/or firmware upgrade?

schuettecarsten, 2024-06-03:
The missing file is /usr/share/nftables.d/ruleset-post/30-pbr.nft.

stangri, 2024-06-03:
That file is re-created on:
* boot
* start
* reload
* restart

schuettecarsten, 2024-06-03:
Ok, I looked into the logs while firewall loads and expected that the file is recreated in this case also.

stangri, 2024-06-03:
> Ok, I looked into the logs while firewall loads and expected that the file is recreated in this case also.

Ah, you're right, It should be re-created on the firewall reload if that file doesn't exist. In that case the `on_firewall_reload` command to procd script turns into `start` command.

schuettecarsten, 2024-06-10:
I also run into the issue that the file is NOT recreated on the boot after a firmeware flash.

stangri, 2024-06-10:
> I also run into the issue that the file is NOT recreated on the boot after a firmware flash, pbr ends in invalid state and needs to be restarted manually to work properly after flash.

What do you mean "also"?

Please elaborate on exact steps of firmware flash.

schuettecarsten, 2024-06-10:
I upload the file to /tmp/firmware.bin and run `sysupgrade -v -u /tmp/firmware.bin` on an SSH console. On the next boot, PBR does not load and shows some "Invalid" errors on its Luci status page. "Reload" fixes this.

stangri, 2024-06-10:
Thanks for your prompt reply. Going from which version of OpenWrt to which version of OpenWrt? What's in the log for pbr after first boot?

schuettecarsten, 2024-06-10:
I just set up a rsyslog server and let you know on my next flash - tomorrow. Then I will also add some screenshots to clarify the issue and my settings.

stangri, 2025-01-17:
Closing due to lack of replies.

closed 2025-01-17 by stangri