mossdef-org/source.mossdef.org #202: [pbr] issue: IPv6, missing default route in VPN table if the default route is not present in main table

egc112, 2024-06-21:
[pbr] issue: IPv6, missing default route in VPN table if the default route is not present in main table

I was experimenting with IPv6 and PBR and ran into some problems.

This is a test router with which I am tinkering a lot, so it might just be gremlins on the router.
```
DL-WRX36, K6.33
pbr 1.1.5-8 running on OpenWrt SNAPSHOT. WAN (IPv4): wan/wan/192.168.21.1. WAN (IPv6): wan6/wan/2a01:cb18:8660:de00::/64.
```

IPv6 works without PBR with a WireGuard tunnel to Mullvad.
I had to do some tweaks to get the default routing via the VPN  right:
```
Network
•	WAN6 > Advanced tab  > Disable Source routing
•	WAN4 set metric to 20
•	WG interface set metric to 10 
Firewall
•	The WG interface needs to be in its one zone
WG firewall zone>  Advanced settings > Enable IPv6 Masquerading

```

With this ipleak.net and traceroute show the use of IPv6 via the tunnel.

Now on to PBR.
When PBR is enabled, tables for WAN and VPN are made with appropriate default routing

The problem arises, for IPv6 only, when I disable default routing by either disabling Route Allowed IPs or disabling default gateway.
IPv4 is working, but in the IPv6 VPN table there is no longer a default route via the VPN

I looked at the code and although I do not grasp all the details there seems to be a difference between IPv4 and IPv6

### IPv4:
```
				if [ -n "$gw4" ] || [ "$strict_enforcement" -ne '0' ]; then
					ipv4_error=0
					if [ -z "$gw4" ]; then
						try "$ip_bin" -4 route add unreachable default table "$tid" >/dev/null 2>&1 || ipv4_error=1
					else
						try "$ip_bin" -4 route add default via "$gw4" dev "$dev" table "$tid" >/dev/null 2>&1 || ipv4_error=1
					fi
# shellcheck disable=SC2086
					while read -r i; do
						i="$(echo "$i" | sed 's/ linkdown$//')"
						i="$(echo "$i" | sed 's/ onlink$//')"
						idev="$(echo "$i" | grep -Eso 'dev [^ ]*' | awk '{print $2}')"
						if ! is_supported_iface_dev "$idev"; then
							try "$ip_bin" -4 route add $i table "$tid" >/dev/null 2>&1 || ipv4_error=1
						fi
					done << EOF
					$($ip_bin -4 route list table main)
EOF
```

As far as I can see it will either add and unreachable or the default route to the routing table and then adds "local" rules to the routing table

### IPv6
The IPv6 code seems to do things differently:
```
				if [ -n "$ipv6_enabled" ]; then
					ipv6_error=0
					$ip_bin -6 rule del table "$tid" >/dev/null 2>&1
					$ip_bin -6 route flush table "$tid" >/dev/null 2>&1
					if { [ -n "$gw6" ] && [ "$gw6" != "::/0" ]; } || [ "$strict_enforcement" -ne '0' ]; then
						if [ -z "$gw6" ] || [ "$gw6" = "::/0" ]; then
							try "$ip_bin" -6 route add unreachable default table "$tid" >/dev/null 2>&1 || ipv6_error=1
						elif "$ip_bin" -6 route list table main | grep -q " dev $dev6 "; then
							"$ip_bin" -6 route add default via "$gw6" dev "$dev6" table "$tid" >/dev/null 2>&1 || ipv6_error=1
							while read -r i; do
[... 550 lines cut ...]
        option name 'Plex/Emby Local Server'
        option interface 'wan'
        option src_port '8096 8920 32400'
        option enabled '0'

config policy
        option name 'Plex/Emby Remote Servers'
        option interface 'wan'
        option dest_addr 'plex.tv my.plexapp.com emby.media app.emby.media tv.emby.media'
        option enabled '0'

config include
        option enabled '0'
        option path '/usr/share/pbr/pbr.user.wg_server_and_client'

config include
        option enabled '0'
        option path '/usr/share/pbr/pbr.user.aws.egc'

config policy
        option name 'fe20'
        option src_addr '98:B8:BC:8B:3F:9E'
        option interface 'wg_mullv_se'

```

stangri, 2024-06-21:
Thank you for the report and suggested patch Erik. I'm in the process of preparing to remove the iptables-related code and reworking the nft code to improve the nft_file_mode support (by adding all elements to the nft set in one command and thus hopefully avoiding the nft set collision errors), so I'll try to review this with the attention it deserves at some point next week.

egc112, 2024-06-22:
Thanks, good idea to remove the iptables-related code as stable and main builds are now nftables

closed 2024-11-12 by egc112