mossdef-org/source.mossdef.org #203: domain policy not resolving the ip in nftset automatically

crushed112, 2024-11-29:
**Describe the bug**

When defining a domain-based policy (dest_addr) in the PBR configuration, the expected behavior is for dnsmasq to dynamically resolve the domain and populate the associated nftset. However, the nftset remains empty, and the policy does not apply unless the IP is manually added to the nftset.

**Your configs**

1. ```/etc/config/dhcp```

```sh
root@bpi-r3:~# cat /etc/config/dhcp
config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option cachesize '1000'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option localservice '1'
	option ednspacket_max '1232'
	option logqueries '1'
	option logfacility '/var/log/dnsmasq.log'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	option force '1'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

config host
	option name 'npi-r2s'
	list mac 'XX:XX:XX:XX:XX:XX'
	option ip '192.168.1.243'
	option leasetime 'infinite'
```

2. ```/etc/config/network```

```sh
root@bpi-r3:~# cat /etc/config/dhcp

config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option cachesize '1000'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
[... 394 lines cut ...]
30000:	from all fwmark 0x10000/0xff0000 lookup pbr_wan
IPv4 table 257 route: default via 10.2.0.2 dev vpnwan_CO_136 
IPv4 table 257 rule(s):
29998:	from all fwmark 0x20000/0xff0000 lookup pbr_vpnwan_CO_136

```
```sh
root@bpi-r3:~# ubus call system board
{
	"kernel": "6.6.61",
	"hostname": "bpi-r3",
	"system": "ARMv8 Processor rev 4",
	"model": "Bananapi BPI-R3",
	"board_name": "bananapi,bpi-r3",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "SNAPSHOT",
		"revision": "r28146-52b6c92479",
		"target": "mediatek/filogic",
		"description": "OpenWrt SNAPSHOT r28146-52b6c92479",
		"builddate": "1732223081"
	}
}
```

crushed112, 2024-11-29:
running  
```sh 
nft add element inet fw4 pbr_vpnwan_CO_136_4_dst_ip_cfg026ff5 { 23.239.16.110 }
```
does make the policy work also

stangri, 2024-11-30:
Hey, thanks for the donation and Happy Thanksgiving!

How much of the https://docs.openwrt.melmac.net/pbr/#AWordAboutBrokenDomainPolicies have you gone thru? For majority of people the last big bullet point in that section (then further divided in sub-bullet points) applies. 

From your configs I'd venture a guess that you're using a pihole (or another resolver) on your LAN, so it is likely that your clients do not make any resolution requests to your router/dnsmasq.

crushed112, 2024-11-30:
Happy thanksgiving and thanks for the fast response! 

Ah I didn't actually read that full section! I saw the first bit and misread it as the router just needs to be able to make DNS requests 

I didn't realize the router actually needs to be the DNS server 

Yes you're right I do have Adguard home separate from my router, does this mean I won't be able to do domain based policies?

Thank you, I guess I didn't make the connection that the policies would actually be enforced when the domains were resolved

stangri, 2024-11-30:
> Yes you're right I do have Adguard home separate from my router, does this mean I won't be able to do domain based policies?

You have a few options on how to make pbr work.

1. Instead of configuring your local devices to use AGH as the resolver, you set the dnsmasq to use AGH as the resolver and let local devices use your router/dnsmasq for resolution. That would make pbr work, but the reports in AGH will show one source of DNS requests -- your router.
2. If you have a router (like Dell Edge 620, which can be bought second hand on ebay for under $100), you can run AGH on your router. AGH can populate ipsets and that option is supported in the (pre-nft) iptables/ipset-flavour of pbr, so there's a hope that in the future AGH versions would also learn to populate nft sets and it can then be used with the regular version of pbr. That way you can keep using AGH for the domain resolution and it would populate nft sets that pbr creates (again, once AGH is nft-aware).

crushed112, 2024-11-30:
Ah option 1 sounds ideal, I could run it on this router too but I'm going to try option 1 first thanks so much I'll close this issue as complete now

closed 2024-11-30 by crushed112

crushed112, 2024-12-02:
@stangri for Option 1 would this be done by setting a DNS forward to my adguardhome's IP in the luci DNS AND DHCP tab?

I think I've done this yet it still doesn't seem to have solved the domain based rules 

Thanks again and sorry for noob question

stangri, 2024-12-02:
I don't use luci much, but I just checked and yes, it would be Network -> DHCP & DNS -> Forwards in the DNS Forwards section. At least on 23.05, I know there were changes to the English text labels, maybe it's DNS Forwardings now.