mossdef-org/source.mossdef.org #204: [pbr] issue: `Installing fw4 nft file` fails with straightforward rule?

posita, 2024-12-30:
**Describe the bug**
A clear and concise description of what the bug is.

**Your configs**

1. ```/etc/config/dhcp```

```sh
# ...
```

2. ```/etc/config/network```

```sh

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'lan5'

config device
	option name 'lan1'
	option macaddr 'cc:28:aa:5d:ac:6a'

config device
	option name 'lan2'
	option macaddr 'cc:28:aa:5d:ac:6a'

config device
	option name 'lan3'
	option macaddr 'cc:28:aa:5d:ac:6a'

config device
	option name 'lan4'
	option macaddr 'cc:28:aa:5d:ac:6a'

config device
	option name 'lan5'
	option macaddr 'cc:28:aa:5d:ac:6a'

config interface 'lan'
	option device 'br-lan.32'
	option proto 'static'
	option ipaddr '192.168.32.1'
	option netmask '255.255.254.0'
	option ipv6 '0'
	option delegate '0'

config device
	option name 'eth1'
	option macaddr 'cc:28:aa:5d:ac:6a'

config interface 'wan'
	option device 'eth1'
	option proto 'dhcp'
	option hostname '*'
	option ipv6 '0'
	option delegate '0'
	option peerdns '0'
[... 434 lines cut ...]
pbr chains - policies
        chain pbr_forward { # handle 59
        }
        chain pbr_input { # handle 60
        }
        chain pbr_output { # handle 61
        }
        chain pbr_postrouting { # handle 63
        }
        chain pbr_prerouting { # handle 62
        }
        chain pbr_dstnat { # handle 58
        }

pbr chains - marking

pbr nft sets

IPv4 table 256 route: default via XXX.XXX.XXX.XXX dev eth1
IPv4 table 256 rule(s):
30000:  from all fwmark 0x10000/0xff0000 lookup pbr_wan
IPv4 table 257 route: default via YYY.YYY.YYY.YYY dev ovpnc0
IPv4 table 257 rule(s):
29998:  from all fwmark 0x20000/0xff0000 lookup pbr_vpn
```

posita, 2024-12-30:
This problem appears to go away with 1.1.8, but after upgrading (and restarting my configuration from scratch), PBR no longer appears to work reliably. I will file another issue for that problem. (I don't know if this is related to #193.)

closed 2024-12-30 by posita

posita, 2024-12-30:
> after upgrading (and restarting my configuration from scratch), PBR no longer appears to work reliably

Following up on this, I got hit by this: https://forum.openwrt.org/t/pbr-not-working-with-some-domains/183426

So `ifconfig.me` (and quite possibly `ifcfg.me`) don't appear to work with PBR?

stangri, 2024-12-30:
I'm struggling to understand why validation script did not complain about:
```
	option resolver_set 'dnsmasq.ipset'
```
for the nft-capable version, but I suspect the error comes from the fact that no nft set was created due to incompatible dnsmasq instruction.

> So `ifconfig.me` (and quite possibly `ifcfg.me`) don't appear to work with PBR?

Most likely the user error. Please peruse the README.

posita, 2024-12-30:
> I'm struggling to understand why validation script did not complain about:
> 
> ```
> 	option resolver_set 'dnsmasq.ipset'
> ```
> 
> for the nft-capable version, but I suspect the error comes from the fact that no nft set was created due to incompatible dnsmasq instruction.

Apologies for not being clear. I omitted that line after upgrading to 1.1.8, so perhaps that would have fixed things for 1.1.6 as well? Not sure, but I'm not sure it matters, since it's resolved. :+1:

posita, 2024-12-30:
> > So `ifconfig.me` (and quite possibly `ifcfg.me`) don't appear to work with PBR?
> 
> Most likely the user error. Please peruse the README.

I'm the second person to hit that, and https://forum.openwrt.org/t/pbr-not-working-with-some-domains/183426 doesn't indicate any fix. With respect to the README, which part do you think would be helpful? It's not clear to me why that should silently not work when other domains in the same entry do. Also, those docs appear to be for 1.1.6, not 1.1.8? It's not clear. Can you point me to the right place? Thanks!