mossdef-org/source.mossdef.org #206: [pbr] wish:

usr97629238, 2025-01-11:
Could you please add support for multiple-IP (secondary IP) assignment per interface?
They are supported by the OpenWrt per se, what is needed is for **pbr** to pick them up as different interfaces.

**Is your feature request related to a problem? Please describe.**
Sometimes routing is needed not only via internal OpenWrt VPNs, but also via external devices/VPNs. And these secondary IPs can each have it own gateway ip.

**Describe the solution you'd like**
**pbr** should pick up each ip as a different interface, like lan-1, lan-2.

**Describe alternatives you've considered**
The alternatives would be adding ipip or other tunnels, however that adds an unneeded overhead.

**Additional context**
You add them like:
```
# /etc/config/network:

## primary
config interface 'lan'
option device 'br-lan'
option proto 'static'
option ipaddr '10.0.0.13'
option netmask '255.255.255.0'
option gateway '10.0.0.3'  

## secondary
config interface 'lan2'
option proto 'static'
option device 'br-lan'
option ipaddr '10.0.0.14'
option netmask '255.255.255.0'
option gateway '10.0.0.4'

# Add tables
echo "201 table1" >> /etc/iproute2/rt_tables
echo "202 table2" >> /etc/iproute2/rt_tables

# Add routes
ip route add 10.0.0.0/22 dev br-lan table table1
ip route add default via 10.0.0.3 dev br-lan table table1

ip route add 10.0.0.0/22 dev br-lan table table2
ip route add default via 10.0.0.4 dev br-lan table table2

# Add rules
ip rule add from 10.0.0.13 lookup table1
ip rule add from 10.0.0.14 lookup table2
```

And everything works as intended:
```
ip a
...
5: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether XXXXXXXXXX brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.13/22 brd 10.0.0.255 scope global br-lan
        valid_lft forever preferred_lft forever
    inet 10.0.0.14/24 brd 10.0.0.255 scope global secondary br-lan
        valid_lft forever preferred_lft forever

ping -I 10.0.0.13 debian.org
PING debian.org (151.101.2.132) from 10.0.0.13: 56 data bytes
64 bytes from 151.101.2.132: seq=0 ttl=58 time=165.137 ms
64 bytes from 151.101.2.132: seq=1 ttl=58 time=166.281 ms
64 bytes from 151.101.2.132: seq=2 ttl=58 time=166.198 ms 

ping -I 10.0.0.14 debian.org
PING debian.org (151.101.130.132) from 10.0.0.14: 56 data bytes
64 bytes from 151.101.130.132: seq=0 ttl=59 time=31.068 ms
64 bytes from 151.101.130.132: seq=1 ttl=59 time=30.889 ms
64 bytes from 151.101.130.132: seq=2 ttl=59 time=30.064 ms  

```

stangri, 2025-01-11:
Sorry, what exactly do you want `pbr` to do in this case?

usr97629238, 2025-01-11:
The same thing as with internal VPNs - rule based routing between them. Same like when for example traffic to siteA goes via tun_internal_openvpn_1 and to siteB via wan, that would allow to have additional rules to route for example traffic to siteC via tun_external_vpn2 and traffic to siteD via tun_external_vpn3.

stangri, 2025-01-12:
Have you tried adding those interfaces to the supported_interface lists?

stangri, 2025-01-17:
Closing due to lack of replies.

closed 2025-01-17 by stangri