mossdef-org/source.mossdef.org #207: [pbr] issue: pbr no longer re-routes (or just outright blocks) policy domains

posita, 2025-01-14:
**Describe the bug**

Follow-up to #193. I can no longer get `pbr` to work reliably at 1.1.8-r4. It either has no affect or ends up blocking policy addresses/domains.

**Your configs**

1. ```/etc/config/dhcp```

```sh
config dnsmasq
	option domainneeded '1'
	option boguspriv '1'
	option filterwin2k '0'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'local'
	option expandhosts '1'
	option nonegcache '0'
	option cachesize '1000'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option nonwildcard '1'
	option localservice '1'
	option ednspacket_max '1232'
	option filter_aaaa '0'
	option filter_a '0'
	option sequential_ip '1'
	option confdir '/tmp/dnsmasq.d'

config dhcp 'lan'
	option interface 'lan'
	option start '128'
	option limit '123'
	option leasetime '12h'
	option dhcpv4 'server'
	option ra 'disabled'
	option ra_flags 'none'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

config dhcp 'iot'
	option interface 'iot'
	option start '128'
	option limit '123'
	option leasetime '12h'

config dhcp 'guest'
	option interface 'guest'
	option start '128'
	option limit '123'
	option leasetime '12h'
	option force '1'
	list dhcp_option '6,9.9.9.9'
	list dhcp_option '6,149.112.112.112'

config domain
	option name 'modem'
	option ip '192.168.100.1'
[... 398 lines cut ...]
	}

pbr nft sets
	set pbr_wan_4_dst_ip_cfg086ff5 { # handle 2751
		type ipv4_addr
		flags interval
		auto-merge
		comment "lan-wan"
		elements = { 104.21.54.91, 172.67.168.106 }
	}

dnsmasq sets
nftset=/ifconfig.co/4#inet#fw4#pbr_wan_4_dst_ip_cfg086ff5 # lan-wan

pbr tables & routing
IPv4 table 256 pbr_wan route:
default via <ISP gateway> dev eth1 
IPv4 table 256 pbr_wan rule(s):
30000:	from all fwmark 0x10000/0xff0000 lookup pbr_wan

IPv4 table 257 pbr_vpn route:
default via <VPN assigned IP> dev ovpnc0 
IPv4 table 257 pbr_vpn rule(s):
29998:	from all fwmark 0x20000/0xff0000 lookup pbr_vpn
```

posita, 2025-01-14:
FYI, these are the **_only_** changes to `/etc/config/pbr` that I made from the default installation of 1.1.8-r4:

``` sh
uci set 'pbr.config.enabled=1'
uci add_list 'pbr.config.supported_interface=vpn'
uci add pbr policy
uci set 'pbr.@policy[-1].name=lan-wan'
uci set 'pbr.@policy[-1].src_addr=192.168.32.0/24'
uci set 'pbr.@policy[-1].interface=wan'
uci set 'pbr.@policy[-1].dest_addr=192.168.100.1 ifconfig.co'
/etc/init.d/pbr enable
/etc/init.d/pbr reload
```

Everything else is as-installed.

posita, 2025-01-15:
More info, if needed:

``` sh
root@router:~# ip route show
0.0.0.0/1 via 10.96.0.1 dev ovpnc0
default via <ISP gateway> dev eth1 proto static src <ISP assigned IP>
10.10.10.0/24 dev lan2 proto kernel scope link src 10.10.10.10 linkdown
10.96.0.0/16 dev ovpnc0 proto kernel scope link src 10.96.0.30
45.134.140.59 via <ISP gateway> dev eth1
128.0.0.0/1 via 10.96.0.1 dev ovpnc0
<ISP subnet>/21 dev eth1 proto kernel scope link src <ISP assigned IP>
192.168.32.0/24 dev br-lan.32 proto kernel scope link src 192.168.32.1
192.168.33.0/24 dev br-lan.33 proto kernel scope link src 192.168.33.1
192.168.47.0/24 dev br-lan.47 proto kernel scope link src 192.168.47.1
root@router:~# ip rule show
0:      from all lookup local
29998:  from all fwmark 0x20000/0xff0000 lookup pbr_protonvpn
30000:  from all fwmark 0x10000/0xff0000 lookup pbr_wan
32766:  from all lookup main
32767:  from all lookup default
root@router:~# cat /var/run/pbr.dnsmasq
nftset=/ipconfig.co/4#inet#fw4#pbr_wan_4_dst_ip_cfg096ff5 # lan-wan
root@router:~# cat /var/run/pbr.nft
#!/usr/sbin/nft -f

add chain inet fw4 pbr_mark_0x010000
add rule inet fw4 pbr_mark_0x010000  mark set mark and 0xff00ffff xor 0x010000
add rule inet fw4 pbr_mark_0x010000 return
add chain inet fw4 pbr_mark_0x020000
add rule inet fw4 pbr_mark_0x020000  mark set mark and 0xff00ffff xor 0x020000
add rule inet fw4 pbr_mark_0x020000 return
add set inet fw4 pbr_wan_4_dst_ip_cfg096ff5 { type ipv4_addr;            auto-merge;                             flags interval;                                 policy performance;                                  comment "lan-wan";}
add rule inet fw4 pbr_prerouting ip saddr { 192.168.32.0/24 } ip daddr @pbr_wan_4_dst_ip_cfg096ff5  goto pbr_mark_0x010000 comment "lan-wan"
add rule inet fw4 pbr_prerouting ip saddr { 192.168.32.0/24 } ip daddr { 192.168.100.1 }  goto pbr_mark_0x010000 comment "lan-wan"
```

stangri, 2025-01-15:
From the service output I can see that:
1. pbr has successfully created a valid nft file with enabled policies translated into nft rules
2. dnsmasq nftset file has been created with the domain name from the enabled policy
3. dnsmasq populated nftset with the IP addresses to which the domain from enabled policy resolves
So looks like policies-related part is working.

Please clarify what happens with your VLAN traffic if:
1. You stop/disable `pbr`?
2. You stop/disable `pbr` and stop the OpenVPN instance?

posita, 2025-01-15:
Ugh. Lost three whole days to this. It turns out there can be no subnet overlap among multiple policies. **_Don't do this_**:

```
config policy
        option name 'first-subnet-only'
        option interface 'wan'
        option src_addr '192.168.32.0/24'
        option dest_addr 'domain1.dom domain2.dom'

config policy
        option name 'first-and-second-subnet'
        option interface 'wan'
        option src_addr '192.168.32.0/24 192.168.34.0/24'
        option dest_addr 'domain3.dom domain4.dom'
```

Do this instead:

```
config policy
        option name 'first-subnet'
        option interface 'wan'
        option src_addr '192.168.32.0/24'
        option dest_addr 'domain1.dom domain2.dom domain3.dom domain4.dom'

config policy
        option name 'second-subnet'
        option interface 'wan'
        option src_addr '192.168.34.0/24'
        option dest_addr 'domain3.dom domain4.dom'
```

I really wish this limitation was properly documented or that I had some help. This was very difficult for me to troubleshoot.

stangri, 2025-01-16:
> FYI, these are the **_only_** changes to `/etc/config/pbr` that I made from the default installation of 1.1.8-r4:
> 
> ```shell
> uci set 'pbr.config.enabled=1'
> uci add_list 'pbr.config.supported_interface=vpn'
> uci add pbr policy
> uci set 'pbr.@policy[-1].name=lan-wan'
> uci set 'pbr.@policy[-1].src_addr=192.168.32.0/24'
> uci set 'pbr.@policy[-1].interface=wan'
> uci set 'pbr.@policy[-1].dest_addr=192.168.100.1 ifconfig.co'
> /etc/init.d/pbr enable
> /etc/init.d/pbr reload
> ```
> 
> All other `pbr` settings are as-installed. I have reviewed and re-reviewed the [footnotes](https://docs.openwrt.melmac.net/pbr/#FootnotesKnownIssues) and the [FAQ](https://docs.openwrt.melmac.net/pbr/#FAQ).

You have previously stated that the above doesn't work. Does it work now?

> It turns out there can be no subnet overlap among multiple policies. I really wish this limitation was properly documented or that I had some help. This was very difficult for me to troubleshoot.

I don't see how this can be a `pbr` limitation.

posita, 2025-01-21:
Okay, I think there are multiple things going on here. I keep encountering issues as I attempt to add entries. Removing them again doesn't seem to help (problems seem to accumulate), but wiping the router and reinstalling, then building back up often helps, which strongly suggest residual side effects. That's the first thing I'm struggling with.

The second thing is that policies are often only partially honored, and results are inconsistent, even over long periods of time. For example, as exits, I now have WAN, WAN2, and a VPN (tun `ovpnc0` via OpenVPN which I keep up all the time). My source addresses come from several different subnets: `192.168.32.0/24` (lan), `192.168.33.0/24` (iot), `192.168.48.0/24` (otherlan), and `192.168.63.0/24` (otherguest). My policies are:

```
# These are often honored (they circumvent my VPN)
config policy
	option name 'lan-wan'
	option interface 'wan'
	option dest_addr 'ip.me speedtest.net speedtest.xfinity.com'
	option src_addr '192.168.32.0/24'

# These are rarely, if ever honored (these almost always go through my VPN)
config policy
	option name 'iot-wan'
	option interface 'wan'
	option dest_addr 'ip.me speedtest.net speedtest.xfinity.com'
	option src_addr '192.168.33.0/24'

# These are often honored (they circumvent my VPN)
config policy
	option name 'lan-wan2'
	option interface 'wan2'
	option dest_addr 'ifconfig.co espn.api.edge.bamgrid.com d2f2ekwwtg17a.cloudfront.net live-mediashield-hulustream-com.akamaized.net hulu.com live-f.hulustream.com fast.com netflix.com nflxext.com nflxvideo.net nflximg.net nflxso.net api.us-east-1.aiv-delivery.net'
	option src_addr '192.168.32.0/24'

# These are rarely, if ever honored (these almost always go through my VPN)
config policy
	option name 'iot-wan2'
	option interface 'wan2'
	option dest_addr 'ifconfig.co espn.api.edge.bamgrid.com d2f2ekwwtg17a.cloudfront.net live-mediashield-hulustream-com.akamaized.net hulu.com live-f.hulustream.com fast.com netflix.com nflxext.com nflxvideo.net nflximg.net nflxso.net api.us-east-1.aiv-delivery.net'
	option src_addr '192.168.33.0/24'

# These are honored
config policy
	option name 'otherlan-wan2'
	option interface 'wan2'
	option src_addr '192.168.48.0/24'

# These are honored
config policy
	option name 'otherguest-wan2'
	option interface 'wan2'
	option src_addr '192.168.63.0/24'
```

Here's what I get when interacting with `pbr`:

``` sh
root@OpenWrt:~# ip route show
0.0.0.0/1 via 10.96.0.1 dev ovpnc0
default via <wan-gw> dev eth1 proto static src <wan-ip> metric 10
default via <wan2-gw> dev lan1 proto static src <wan2-ip> metric 20
10.10.10.0/24 dev lan5 proto kernel scope link src 10.10.10.10 linkdown
10.96.0.0/16 dev ovpnc0 proto kernel scope link src 10.96.0.21
<wan2-datacenter> dev lan1 proto static scope link src <wan2-ip> metric 20
<vpn-pt1> via <wan-gw> dev eth1
<wan2-nm>.0.0/10 dev lan1 proto static scope link metric 20
128.0.0.0/1 via 10.96.0.1 dev ovpnc0
<wan-nm>.0/21 dev eth1 proto static scope link metric 10
192.168.32.0/24 dev br-lan.32 proto kernel scope link src 192.168.32.1
192.168.33.0/24 dev br-lan.33 proto kernel scope link src 192.168.33.1
192.168.34.0/24 dev br-lan.34 proto kernel scope link src 192.168.34.1
192.168.47.0/24 dev br-lan.47 proto kernel scope link src 192.168.47.1
192.168.48.0/24 dev br-lan.48 proto kernel scope link src 192.168.48.1
192.168.63.0/24 dev br-lan.63 proto kernel scope link src 192.168.63.1
<wan2-modem> dev lan1 proto static scope link src <wan2-ip> metric 20
<wan2-svc2> dev lan1 proto static scope link src <wan2-ip> metric 20
root@OpenWrt:~# service pbr reload
Using wan interface (on_start): wan
[... 285 lines cut ...]
 2  10.96.0.1 (10.96.0.1)  38.128 ms  45.826 ms  45.915 ms
 3  <vpn-pt1> ...

traceroute to ip.me (212.102.35.236), 3 hops max, 60 byte packets
 1  * * *
 2  * * *
 3  <vpn-pt1> ...

traceroute to speedtest.net (151.101.194.219), 3 hops max, 60 byte packets
 1  * * *
 2  * * *
 3  <vpn-pt1> ...

traceroute to speedtest.xfinity.com (96.97.11.44), 3 hops max, 60 byte packets
 1  192.168.33.1 (192.168.33.1)  1.087 ms * *
 2  10.96.0.1 (10.96.0.1)  39.698 ms * *
 3  <vpn-pt1> ...

https://icanhazip.com/: <vpn-exit>
https://ifconfig.co/: <vpn-exit>
```

I would hope and expect those two outcomes would be the same.

How can I diagnose these inconsistencies? This is driving me absolutely nuts.

stangri, 2025-08-10:
Sorry forgot to tag it when it got fixed, but the current 1.1.8 and 1.1.9 branches both have the fix in.

closed 2025-10-16 by stangri